mavlink/mavlink-devguide #26: Field Reordering

Hunter522, 2017-11-28:
Went looking through the docs to refresh myself on how field reordering works and didn't find it.

Maybe add what's documented [here](http://qgroundcontrol.org/mavlink/crc_extra_calculation) would be helpful. Perhaps add it to the "Serialization" page?

Not entirely sure what happens in MAVLink2 extended fields, are they factored in reordering or are they separate? Perhaps this should be clarified as well.

hamishwillee, 2017-11-29:
Hi @Hunter522 

Thanks for this. I've created #27 to pull this in.

More generally, I am tasked at some point to either delete or pull everything across from http://qgroundcontrol.org/mavlink/crc_extra_calculation?do=index . Unfortunately I don't have the expertise to know what is useful and what is dross. 
1. Do you think you might be able to help me? Effectively provide the request exactly as you have done here "I think this is missing, material is OK but is missing XYZ, it should go here".
2. Who would you recommend for review of this info and additional information (like the MAVLink2 considerations).

Hunter522, 2017-12-04:
Thanks @hamishwillee.

1. I can help best I can sure.
2. Not entirely sure who is working on each particular part but @LorenzMeier or @tridge seem to be responsible for a lot of the new MAVLink2 work

Also https://github.com/mavlink/mavlink/blob/master/doc/MAVLink2.md may be useful for additional MAVLink2 documentation.

hamishwillee, 2017-12-05:
Thank you. Unfortunately both @LorenzMeier and @tridge are difficult to extract time from for this purpose. Whatever help you can give me will be much appreciated :-)
Regarding the MAVLink2 doc - yes. Created https://github.com/mavlink/mavlink-devguide/issues/29 for tracking.

hamishwillee, 2017-12-14:
Closed by https://github.com/mavlink/mavlink-devguide/pull/27

closed 2017-12-14 by hamishwillee