mavlink/mavlink-devguide #29: Propose merge /doc/MAVLink2.md into developer guide

hamishwillee, 2017-12-05:
Merge: https://github.com/mavlink/mavlink/blob/master/doc/MAVLink2.md
into
https://mavlink.io/en/mavlink_2.html

And then delete the original doc. 

Any objections. Speak now or forever hold your peace!

hamishwillee, 2018-05-04:
As part of this job need to address suggestions here: https://github.com/mavlink/mavlink-devguide/pull/53#issuecomment-386212315

Specifically:
Add most of topic "Using the C implementation" from the MAVLink2.md doc, and that in particular link to https://github.com/tridge/ardupilot/blob/mavlink2-wip/libraries/GCS_MAVLink/GCS_Signing.cpp in the section on Handling message signing.

@julianoes Is there a plan for us to support signing within the core mavlink libraries?

LorenzMeier, 2018-05-10:
Signing is both supported and tested to work. The only thing missing is consistent documentation and making sure that signing support is for both Firmware and QGC on master.

hamishwillee, 2018-05-11:
Thanks @LorenzMeier .  Perhaps the next step is to pull in https://github.com/mavlink/mavlink/blob/master/doc/MAVLink2.md#handling-message-signing so you can highlight what is incorrect. 

Also note that according to that doc "support for variable length arrays" is not implemented - we say it is on the official docs.

LorenzMeier, 2018-05-11:
Pulling in that doc would be useful, yes. On variable length arrays: We are supporting zero-truncation of the largest array. So if a fixed array has zero elements and is the largest == last in the message, it will get truncated. So a "light" version of that is implemented.

hamishwillee, 2018-05-15:
@LorenzMeier 

> On variable length arrays: We are supporting zero-truncation of the largest array. So if a fixed array has zero elements and is the largest == last in the message, it will get truncated. So a "light" version of that is implemented.

I think I understand. Where you say `is the largest == last in the message` do you mean that the message will be organised such that the largest array (?) will be last in the message (and will hence be truncated)

If so, that sounds OK. Very few messages have more than one array, so the only array should be truncated if it has zero elements.

LorenzMeier, 2018-05-15:
Yes, that is how they get organized.

hamishwillee, 2018-06-18:
This was done in #58

closed 2018-06-18 by hamishwillee