mavlink/mavlink-devguide #33: Create contribution guidelines

hamishwillee, 2018-02-09:
We don't have any guidelines to contributing to MAVLink. We also don't have any clarification on how this project works with the upstream version on ArduPilot (we are a fork).

@dagar @julianoes Can we have a discussion about what the process is for updating different parts of the code, tooling, messages?

mrpollo, 2018-02-09:
@hamishwillee I think you got the fork part backward, [APM MAVLink](https://github.com/ArduPilot/mavlink) is a fork of [mavlink/mavlink](https://github.com/mavlink/mavlink)

hamishwillee, 2018-02-09:
@mrpollo Fixed. It makes sense that they are the fork, but for some reason I had it in my head that we'd come from a fork in that project.

mrpollo, 2018-02-09:
I would also like to add a deprecation policy, how to test mavlink changes within PX4 or (insert FC here), @dagar I seem to remember you brought this up on our last strategy meeting can you please expand on this? I'm not doing a great job

hamishwillee, 2018-02-13:
We also need an approval and merge process - otherwise PRs just sit.

amilcarlucas, 2018-02-14:
Hello all, I volunteered to be the contact person between ArduPilot and PX4 regrading mavlink and pymavlink.
So for I have done pull request in both directions that got these four repositories mostly in sync.

* mavlink/mavlink (upstream)
* ardupilot/mavlink (downstream)
* ardupilot/pymavlink (upstream)
* mavlink/pymavlink (downstream)

So far the policy that I used is that the mavlink repositories contain more edgy and untested WIP features. And the ardupilot repositories contain stable, tested, and reviewed code (WIP is not allowed here)

I plan to continue to create PRs and integrate the code in both directions Mavlink->Ardupilot and Ardupilot -> Mavlink

@LorenzMeier  But the current status is good, as soon and mavlink/pymavlink#6 gets merged ;)

julianoes, 2018-02-14:
@amilcarlucas that's great, Thanks for stepping up to do this and help!

hamishwillee, 2018-02-15:
As part of the contribution guide we need to approve channels for discussion. As per https://github.com/mavlink/mavlink-devguide/pull/38/files#r168386782 , where do MAVLink discussions take place, and in particular for contributors?
- I assume that it is slack, and that the google mailing list is closed
- Is the PX4 devcall where MAVLink stuff gets discussed for "contributing"?

hamishwillee, 2018-02-23:
Closing because basic contribution guide has been added: https://mavlink.io/en/contributing/contributing.html

I think further discussion on this should take place when their is an agreed maintainer.

closed 2018-02-23 by hamishwillee