mavlink/mavlink-devguide #61: Doc Bug: Protocols · MAVLink Developer Guide

DavidIngraham, 2018-06-05:
I'm a bit confused about the over-the-wire protocol for MAVLink 2.

The documentation on this page indicates that MAVLink 2 messages with target sys/comp IDs will include the targets as part of the header, and will not sort them with the payload fields. (At least, that's how I'm interpreting the 'Mavlink 2 Packet Format' section)

However, looking through the generated C Library, it appears that the targets are treated as normal payload fields, and are not guaranteed to be at the beginning of the payload. For example, here is the C implementation of the VIDEO_STREAM_SETTINGS message, a MAVLink 2 only message.
https://github.com/mavlink/c_library_v2/blob/master/common/mavlink_msg_set_video_stream_settings.h

Perhaps a note should be added to clarify whether the targets are sorted as part of the payload, and whether or not they are included in the payload length.

Bug Page: [Protocols · MAVLink Developer Guide](https://mavlink.io/en/protocol/overview.html?q=)

LorenzMeier, 2018-06-05:
You are correct. These fields are not in the header, but rather are indexed. You can pull them out using the associated helper functions.

DavidIngraham, 2018-06-06:
@hamishwillee From what I can tell, the best course of action would just be to remove the target sys/comp id lines from the mavlink 2 packet format, and increase the payload size to 255.

However, I don't know if these lines were trying t communicate something else which would need to be documented more appropriately if they were removed

hamishwillee, 2018-10-29:
FYI Closing. This was fixed as suggested a few months ago - thanks @DavidIngraham .

At some point I will update the [C API](https://mavlink.io/en/mavgen_c/) docs with info about the helpers.

closed 2018-10-29 by hamishwillee