mavlink/mavlink-devguide #83: General questions for confirmation

hamishwillee, 2018-07-18:
Confirm:
1.  target_sysid and target_compid are not part of the standard packet frame - they are part of the message for those messages that need them
    - the [Protocol overview](https://mavlink.io/en/protocol/overview.html#mavlink-2-packet-format) indicates incorrectly that they are part of the frame.
    - This has propagated into the [Serialisation Topic](https://mavlink.io/en/guide/serialization.html#mavlink-2-packet-format)
 1. Mavlink "core" maintains a sequence number for current sent messages (current_tx_seq). This is added to every message sent and then iterated. It also maintains current received message (current_rx_seq) and stores the next expected message (I think) which is current_rx_seq+1.
   - BUT the protocol at high level does not do any resending or packet order checking. This is all done in the sub-protocol implementations right?

hamishwillee, 2018-07-19:
`target_sysid` and `target_compid`  - are they part of standard frame in MAVLink 2? 

- `target_sysid` and `target_compid` appear not to be part of the standard packet frame - they are part of the message for those messages that need them:
- the [Protocol overview](https://mavlink.io/en/protocol/overview.html#mavlink-2-packet-format) indicates incorrectly that they are part of the frame.
- This has propagated into the [Serialisation Topic](https://mavlink.io/en/guide/serialization.html#mavlink-2-packet-format)

UPDATE: This requires more review and docs update, but the short answer is that this is not "wrong". MAVLink 2 attempts to highlight these more clearly and provides better support for implementing these fields.

Specifically these are "optional" fields - they are only in the packet if defined in a point-to-point message. They are in the message payload. However the behaviour is standardised (from here needs checking). They are ALWAYS in the first two slots of the packet area for the message. 

The tool provides generic ways to access this info in the "top level" - ie without having to inspect the packet (e.g. in [mavlink_types.h](https://github.com/mavlink/c_library_v2/blob/master/mavlink_types.h#L284-L291)
```
/*
  entry in table of information about each message type
 */
typedef struct __mavlink_msg_entry {
	uint32_t msgid;
	uint8_t crc_extra;
	uint8_t msg_len;
	uint8_t flags;             // MAV_MSG_ENTRY_FLAG_*
	uint8_t target_system_ofs; // payload offset to target_system, or 0
	uint8_t target_component_ofs; // payload offset to target_component, or 0
} mavlink_msg_entry_t;
```

Need to capture this.

hamishwillee, 2018-07-19:
Does MAVLink protocol at high level does do any resending or packet order checking?
- My understanding is that the Mavlink "core" maintains a sequence number for current sent messages (current_tx_seq). This is added to every message sent and then iterated. It also maintains current received message (current_rx_seq) and stores the next expected message (I think) which is current_rx_seq+1.
- But actual checking of sequence numbers is done in the sub-protocol implementations right?

hamishwillee, 2018-07-23:
Is missionlib still useful - do we have a strategy for protocol examples?
See https://github.com/mavlink/mavlink/issues/954

If it is useful then the doc should be linked from the mission and parameter sections

hamishwillee, 2018-08-02:
What general requirements does MAVLink put on resend rates/timeouts for [AUTOPILOT_VERSION](https://mavlink.io/en/messages/common.html#AUTOPILOT_VERSION),  [HEARTBEAT](https://mavlink.io/en/messages/common.html#HEARTBEAT), [SYS_STATUS](https://mavlink.io/en/messages/common.html#SYS_STATUS)

[Old docs](https://github.com/mavlink/mavlink-devguide/pull/79) indicate they should be sent 
- at least rate of 0.1 Hz 
- repeatedly independent of the system state.

Questions:
- Is that correct?
- Are these hard rules or "recommendations"?
- Do the autopilots/APIs/GCS have other specific expectations/behaviours. For example, I know that DroneKit sends heartbeat at 1Hz.
- The above covers what autopilots do. What does a GCS or API have to send? - HEARTBEAT (obviously). What else?

hamishwillee, 2018-08-02:
Is *Message Profile* information useful (ie minimum set, for parameters etc?)

**NOTE** I tend to think that it probably should be in the subprotocol/microservice section) for each service.

Below is docs on profile from old docs.

----
```
The minimum message set is intended for barebones operation.

### Autopilot TX

* [HEARTBEAT](../messages/common.md#HEARTBEAT)
* [SYS_STATUS](../messages/common.md#SYS_STATUS)
* [AUTOPILOT_VERSION](../messages/common.md#AUTOPILOT_VERSION)
* [ATTITUDE_QUATERNION](../messages/common.md#ATTITUDE_QUATERNION)
* [GLOBAL_POSITION_INT](../messages/common.md#GLOBAL_POSITION_INT)
* [GPS_RAW_INT](../messages/common.md#GPS_RAW_INT)
* [RC_CHANNELS](../messages/common.md#RC_CHANNELS)
* [BATTERY_STATUS](../messages/common.md#BATTERY_STATUS)
* [HIGHRES_IMU](../messages/common.md#HIGHRES_IMU)
* [STATUSTEXT](../messages/common.md#STATUSTEXT)

### GCS TX

* [HEARTBEAT](../messages/common.md#HEARTBEAT)
* [MANUAL_CONTROL](../messages/common.md#MANUAL_CONTROL)

## Parameter Protocol Profile

In addition to the minimum message profile, the parameter profile requires handling of these messages:

### Autopilot TX

* [PARAM_VALUE](../messages/common.md#PARAM_VALUE)

### GCS TX

* [PARAM_REQUEST_READ](../messages/common.md#PARAM_REQUEST_READ)
* [PARAM_REQUEST_LIST](../messages/common.md#PARAM_REQUEST_LIST)
* [PARAM_SET](../messages/common.md#PARAM_SET)

## Mission Protocol Profile

In addition to the minimum message profile, the mission profile requires handling of these messages:

### Autopilot TX

* [MISSION_ITEM](../messages/common.md#MISSION_ITEM)
* [MISSION_REQUEST_LIST](../messages/common.md#MISSION_REQUEST_LIST)
* [MISSION_CLEAR_ALL](../messages/common.md#MISSION_CLEAR_ALL)
* [MISSION_ITEM_REACHED](../messages/common.md#MISSION_ITEM_REACHED)
* [MISSION_ACK](../messages/common.md#MISSION_ACK)

### GCS TX

* [MISSION_ITEM](../messages/common.md#MISSION_ITEM)
* [MISSION_ITEM_INT](../messages/common.md#MISSION_ITEM_INT) (preferred)
* [MISSION_REQUEST](../messages/common.md#MISSION_REQUEST)
* [MISSION_CURRENT](../messages/common.md#MISSION_CURRENT)
* [MISSION_COUNT](../messages/common.md#MISSION_COUNT)
* [MISSION_ACK](../messages/common.md#MISSION_ACK)

## Robotics / Companion Computer Protocol Profile

In addition to the minimum message profile, the robotics profile requires handling of these messages:

### Autopilot TX

* [SYSTEM_TIME](../messages/common.md#SYSTEM_TIME)
* [ATTITUDE_QUATERNION_CO](../messages/common.md#ATTITUDE_QUATERNION_COV)
* [GLOBAL_POSITION_INT_COV](../messages/common.md#GLOBAL_POSITION_INT_COV)
* [LOCAL_POSITION_NED_COV](../messages/common.md#LOCAL_POSITION_NED_COV)
* [ATTITUDE_TARGET](../messages/common.md#ATTITUDE_TARGET)
* [POSITION_TARGET_LOCAL_NED](../messages/common.md#POSITION_TARGET_LOCAL_NED)
* [POSITION_TARGET_GLOBAL_INT](../messages/common.md#POSITION_TARGET_GLOBAL_INT)

### Companion Computer TX

* [COMMAND_INT](../messages/common.md#COMMAND_INT)
* [SET_ATTITUDE_TARGET](../messages/common.md#SET_ATTITUDE_TARGET)
* [SET_POSITION_TARGET_LOCAL_NED](../messages/common.md#SET_POSITION_TARGET_LOCAL_NED)
* [SET_POSITION_TARGET_GLOBAL_INT](../messages/common.md#SET_POSITION_TARGET_GLOBAL_INT)
```

LorenzMeier, 2018-08-02:
These would make sense per microservice, yes. That would be quite educational for implementers.