mavlink/mavlink-devguide #87: Find place and document "Control Interface"

hamishwillee, 2018-07-23:
Currently we use [Protocols · MAVLink Developer Guide](http://localhost:4000/en/protocol/overview.html) to group sub protocols (ie Mission, Command, etc). As part of documenting the protocol this also sensibly groups messages.

However we don't have any similar spot to group "interfaces" - logical sets of messages without any special protocol over/above the command protocol (say). For example Offboard (Control) messages.

Possible option is to make the Protocols section - Protocols & Interfaces.

hamishwillee, 2018-07-23:
Near empty doc on interface from old site:

> ## Control Interface
> In contrast to the parameter or waypoint interface these messages are not part of a formal protocol, but serve all the same purpose of providing controller set points to the autopilot. The most common application of this interface is an onboard companion computer.

LorenzMeier, 2018-07-23:
The offboard interface has a statemachine as well, so in fact it’s the same. Let’s call all of them Micro services and work on documenting all state machines.

hamishwillee, 2018-07-23:
That is a good idea. I will move towards that.

hamishwillee, 2018-07-23:
Though I don't love "microservice" either. I tend to think of a service as the implementation of a protocol. So what we have here is a sub-protocol, as documented.