mavlink/mavlink-devguide #98: How does burst mode work for FTP?

hamishwillee, 2018-08-02:
How does Burst ReadFile operations/streaming work?

I know that for normal reads, the file is read in chunks, with an ACK  before each new chunk is sent. My guess is that for `BurstReadFile` the file is chunked and sent without waiting for a request - so basically you just send a stream of ACKs, with sequentially incremented sequence numbers. When you've sent everything you send the last ACK with the burst mode complete field set.

1. Is that correct?
2. Presumably the recipient is in burst mode, so it is sitting waiting for a bunch of ACKs and re-assembles the message - correct?
3. How is the recipient expected to cope with missing packets - it looks like after the burst is complete perhaps the recipient individually requests the missing packets - does it use a normal read/ACK process for that?

hamishwillee, 2022-02-16:
@bkueng @dayjaby This is a post I created back in 2018. Is this your understanding of burst mode? If so I can take a first shot at docs.

hamishwillee, 2022-02-23:
@bkueng Thanks! Is this correct then?

[[image]](https://mermaid-js.github.io/mermaid-live-editor/edit#pako:eNqtlE1r3DAQhv_KoNMuOJBATy4baDdJD4UWuoUe6lIUe7QWlUeuNGJJQ_57x7K9ZBsHQqgvtqR5Z575sO5V7RtUpYr4OyHVeGX1PujubUUgT68D29r2mhg-bHdPN6-CJxy3P3lGCHbfMngzWJfwuUcCYx2C8QFuU4gMAXVjaT9qxOrs8jI7KQHeDwZf5PxGJCtoNOvv5z82vea2gGj_4MYhrYblGtajgyw9Ex853rvtxxVEjNF6mhRviuwnKweSozLjOjSZdiLIL4gsjF3MOrAEdZvoV4SD5XbM4Wftu94h4-Z8ieI0jRnnX2UhcU1Elq9jarfJGAzr4pj6tPEfotBLoyzV5VsrfZydSmGRGnBamplLs1iZi1czXzzq9QtQTyducA6a7qCz4pz2c_NSXsyhQfNUl6UxfAKYeeZKPjt4p3P3Kvqt8xHnAYbDUPVD64U3j2L-kWrvHNaMzRL5VwydJc24G13MTM8yy4EqVCcybRu5Be4Hw0pxix1WqpTPBo1OjitV0YOYpl5Q8Lqx7IMqjXYRC6UT-90d1arkkHA2mm6SyerhLzelfU8)

A few assumptions (reasonable)?
1. The server decides the size of the chunks it wants to stream during burst, but the client (GCS) gets to decide the size it wants when filling in chunks. I assume it will normally use the maximum size
2. The client needs to track the missing data chunks based on the offsets and data sent.
3. There are timeouts for "not got a new packet" while waiting for a new burst payload.

bkueng, 2022-02-23:
Correct, looks good.

- You might want to add another `BurstReadFile` for clarity, or at least mention there can be multiple (there's also an offset specified)
- Can you use client and server (or similar) instead of GCS and Drone, to be more generic?

hamishwillee, 2022-02-23:
Thanks @bkueng 
Where do you mean add an extra burst read? I've added three, showing the offset. I'll also add offset to the last one
[image]

hamishwillee, 2022-02-24:
PS @bkueng Conversations continues on PR: #445
That might need modification pending your answer to question above.

bkueng, 2022-02-24:
> Where do you mean add an extra burst read? See the ones marked up below (I have updated the missing offset marked in circle above).

After the red arrow

hamishwillee, 2022-02-24:
@bkueng That's why I am confused. At this point burst_complete=1. Are you saying that even after complete the server will send BurstReadFile? If so, I presume that it specifies size=0 and an empty data buffer. How many times?

bkueng, 2022-02-25:
`burst_complete=1` does not mean `transfer complete`. A server can split a transfer into multiple bursts. Let's discuss directly if not clear.

PR #445 (2022-03-09): FTP: Add docs for BurstReadFile
Adds docs explaining how burstreadfile works. 

Fixes #98
closed 2022-03-09 by hamishwillee