mavlink/mavlink-devguide #103: See why common.xml doc is so heavyweight

hamishwillee, 2018-08-08:
Or slow to load. Benchmarks:
* https://mavlink.io/en/messages/common.html
* mavlink.org/messages/common

hamishwillee, 2018-08-09:
OK, I have made two changes which should percolate through. 
1. I have added a minifier, which reduces the size of common.xml by about 30% (from 1Mb to about 700K- significant over slow link). 
1. I've further stripped out all the unnecessary html and class info from the HTML in https://github.com/mavlink/mavlink/pull/974 . Note that this change isn't merged yet.

This is the minimum that the file can be reduced to without reducing size.

Re-benchmarking gives:

Action | [mavlink.io](https://mavlink.io/en/messages/common.html) |  [mavlink.org](http://mavlink.org/messages/common)  |  [New mavlink.io](https://hamishwillee.gitbooks.io/ham_mavdevguide/content/v/f2fb9908f7e2245a687af97bf0eaa6fca09020a0/en/messages/common.html)
--- | --- | --- | ---
common | 944Kb / 1.64s | 68.8Kb / 2.45s | 445 / 1s
transferred | 1.54Mb | 69.4kb | 987Kb
finish | 8.1s | 5.77s | 8
DomContentLoaded | 3.23 | 2.25 | 2.62
load | 5.41s | 3.14 |  5.42

Because gitbook.com is a better served server we now actually get that file a lot faster than mavlink.org, even though it is still much bigger. The

hamishwillee, 2018-08-09:
What else can be done/why is this so much bigger?

The reason that  mavlink.io is is so much bigger than mavlink.org is
- The HTML is better - in contains ids for every element, not just a few of them. This makes it better and easier to link.
- The HTML includes an inline Table of Contents. This allows you to quickly preview/browse messages. This is quite large.

We could consider removing these, but IMO they add real value, and we now have similar performance.

amilcarlucas, 2018-08-09:
This is really good work that you have been doing here.
But ... for the people still using makvlink.org, can you please add the units and enum columns?

hamishwillee, 2018-08-09:
Hi @amilcarlucas,
No, because that is going to be shut down at some point. I don't want to encourage people to rely on it.

amilcarlucas, 2018-08-10:
At the moment it contains NO units whatsoever. That confuses new AND old users that happen to stumble with those pages.

hamishwillee, 2018-08-10:
I've made the note up the top of the page bright red and stated the limitations: https://github.com/mavlink/mavlink/pull/975
So if they stumble on the doc they have a good chance of finding the right docs.

hamishwillee, 2018-08-16:
Closing this. New site optimised, and further improvement difficult without sacrificing useful features.

closed 2018-08-16 by hamishwillee