mavlink/mavlink-devguide #144: Negotiating versions with signed systems.

hamishwillee, 2018-12-05:
Questions: Must all connected entities share the same signature/protocol version?

[MAVLink Versions](https://mavlink.io/en/guide/mavlink_version.html#negotiating_versions) isn't very clear on this because it talks in terms of *links* and just two systems. It doesn't cover the case explicitly where you might have a link to a wifi network that has systems that support just MAVLink 1, MAVLink 2, and MAVLink 2 with signing. 

Here are some assumptions that would be good to confimr:
1. When we say "link" we mean "an interface to a mavlink network" 
   - The link might have multiple systems on it.
   - This is just a serial port that might go over a cable or to wifi. 
   - "link" does not imply a connection between just two systems.
   - Channel and link are synonymous in the C library.
 2. AFAIK The protocol does not mandate that an implementation must always send/receive just one protocol over a link.
   - Do the implementations? I.e. Is the protocol version set for each link by the C library, or can it be set on a per-message basis?
     - If it is sent per-link (as I suspect), the implication is that if you have any MAVLink 1 system on the link you want to connect to then you must stick to MAVLink 1.
3. AFAIK signing is configured per link (both in standard and implementations). 
   - The implication is that if a system sets up a link for signing then it can only connect to signed systems.
   - A GCS that is set up for signing would similarly have to use signatures. That means that you can't have a GCS connect to both signed systems and unsigned or MAVLink 1 systems on the same link.
     - Can a GCS have multiple links to the same interface? ie could you have a GCS have to links to the same network port, and use one for secure traffic and the other for unsecure traffix. 
   
If the above assumptions are true, then each link will have to be reserved for a single protocol/signature config because the links can only be configured to use one protocol/signature at a time.

hamishwillee, 2018-12-05:
@julianoes Is this something we could chat about/you know about? In particular I'm still confused as to whether [channels](https://mavlink.io/en/mavgen_c/#streams) and links are synonymous, and whether settings like mavlink version are set on a per link or a per system basis.

julianoes, 2018-12-06:
> 1. When we say "link" we mean "an interface to a mavlink network" 

I would say a link is the connection between two nodes. And a node is a mavlink instance with a unique sysid and compid combination.
And therefore a system would be multiple nodes with the same sysid.

> - "link" does not imply a connection between just two systems.

It means a physical connection but doesn't mean the two nodes are actually connected properly and able to communicate.

> - Channel and link are synonymous in the C library.

A channel is the number you choose when parsing mavlink. From my understanding each "channel" gives you a buffer. And it probably makes sense to use one channel/buffer per link.

> AFAIK The protocol does not mandate that an implementation must always send/receive just one protocol over a link. 

Good question, I have no idea. 
> - Do the implementations? I.e. Is the protocol version set for each link by the C library, or can it be set on a per-message basis?

I think if you use the v1 headers, than you send mavlink 1 and if you use v2 headers it's mavlink 2 but I'm not even sure there.

> - If it is sent per-link (as I suspect), the implication is that if you have any MAVLink 1 system on the link you want to send to then you must stick to MAVLink 1 (you should still be able to receive its messages)

Presumably. You're using the word link in the sense of network here I think. But I don't actually understand how the MAVLink backward compatibility works :confused:.

> 3. AFAIK signing is configured per link (both in standard and implementations). 

Per node, yes.

> - The implication is that if a system sets up a link for signing then it can only connect to signed systems.

If "connect to" means "be able to parse messages from" then yes, I think so.

> - A GCS that is set up for signing would similarly have to use signatures. That means that you can't have a GCS connect to both signed systems and unsigned or MAVLink 1 systems on the same link. 

I mean you could in theory send MAVLink 1 and MAVLink 2 (even signed) on the same link (e.g. a UART or UDP port) by just sending each message two times, once with one library and once with the other. It would be tedious but not impossible and then the receives would just parse whatever they understand. So I think yes it's possible but cumbersome.

> If the above assumptions are true, then each link will have to be reserved for a single protocol/signature config because the links can only be configured to use one protocol/signature at a time.

I would phrase it this way: two nodes are able to talk to each other if they talk the same protocol/signature and they both need to be configured. 

> And a signed system will always broadcast a signed heartbeat. There is no negotiation - the system will only talk to other correctly set up and compatible systems. Correct?

I don't know how the negotiation is supposed to work which is the reason why I'm still assuming that signed MAVLink does not actually work. I still haven't seen it in practice, so I'm doubtful until proven wrong.

The info I have is the Signing section in [Some proposals for MAVLink 2](https://docs.google.com/document/d/1XtbD0ORNkhZ8eKrsbSIZNLyg9sFRXMXbsR2mp37KbIg/edit#heading=h.kxqpon38t2e4).
From what I read there is no negotiation but you rather have to set the same key on both sides. Additionally, from what I remember, both sides have to use the same time and that's where it fell apart in tries that I have seen. Maybe it would work if you ground station has a GPS or NTP time, and the drone has GPS time as well.

hamishwillee, 2018-12-07:
Thanks!

>> When we say "link" we mean "an interface to a mavlink network"

> I would say a link is the connection between two nodes. And a node is a mavlink instance with a unique sysid and compid combination.
> And therefore a system would be multiple nodes with the same sysid.

In most cases the docs consider the case of a mavlink network of two nodes, so link as "a connection between two nodes" is fine. 
The problem with the definition is that it doesn't appear to reflect the reality of the C API. Specifically that for mavlink version, signature etc appear to be stored for the physical connection and the word "link" is used to refer to this connection. That means that every node on the physical connection (e.g. wifi) shares those settings.

hamishwillee, 2018-12-07:
>> Do the implementations? I.e. Is the protocol version set for each link by the C library, or can it be set on a per-message basis?

> I think if you use the v1 headers, than you send mavlink 1 and if you use v2 headers it's mavlink 2 but I'm not even sure there.

No, the MAVLink 1 headers are the mavlink 1 only library, but the MAVLink 2 library supports both versions of the protocol and which you use is specified on a per-channel basis: https://mavlink.io/en/mavgen_c/#sending-and-receiving-mavlink-1-packets

That is why I think that anything on a particular port must use the same settings/channel.

Perhaps I just need to adjust my terminology. Channel means "the physical interface I pump data out of" while "link" means a connection between two systems. If your link is via cable they are synonymous.

hamishwillee, 2018-12-07:
@julianoes OK, so I am now pretty sure I have it. Now updated in the top post: https://github.com/mavlink/mavlink-devguide/issues/144#issue-387986028

MAVLink version/signing are set up per channel (physical interface), and not per node/node connection in our C and Pymavlink libraries. The only question is whether that is a requirement of the standard or just a limitation of the implementation. I suspect the later.

julianoes, 2018-12-07:
> No, definitely per channel: https://mavlink.io/en/mavgen_c/message_signing_c.html#enabling_signing_channel

Right, I think we mean the same thing. What I mean is both nodes need to have it configured to be able to talk.

> The only question is whether that is a requirement of the standard or just a limitation of the implementation. I suspect the later.

I would say so too, it's a limitation of the implementation.

hamishwillee, 2018-12-10:
Thanks for that Ben. Yes, that is pretty much the reverse engineering I did and with the same result: https://github.com/mavlink/mavlink-devguide/issues/144#issue-387986028

IMO the simple answer that generally we'd expect all entities that need to communicate on a network that supports signing to be signed, and to be signed using the same key. 
If you wanted to support unsigned systems, MAVLink 1 systems, or systems that have a different signature that is perfectly possible, but you need to use different channels on your GCS for talking to these. 

A more rambling and pendantic answer below ....

----

Because I am very pedantic:
- we can only say "where multiple autopilots are communicating with a single GCS via the same channel" (or "via the same port"). We can't refer to "connection" in this context because that means a node that is regularly accepting another nodes HEARTBEAT message. "Transmission space" to me means "network".
  - I tend to think of channel as port. That helps because it makes it clear that we don't have a pipe between the two nodes where the pipe has settings. Instead we have a bunch of nodes with their own own outflow/inflow pipe (and settings) into the network. 
- The "result" for different secret keys and versions might be the same, but IMO this is for different reasons. So they should be discussed separately.
- Your analysis is limited to the implementations we have. As discussed above, it is an implementation detail that ties the key and MAVLink version to the channel and not to the node-node connection. So the following discussion might not apply for all implementations.

In any case, I think your questions are answered by https://github.com/mavlink/mavlink-devguide/issues/144#issue-387986028 up the top and by your summary:
1. No you cannot support multiple secret keys via the *same* channel in the current implementations. 
   - All recipients/channels that get a message (e.g. heartbeat) must have that key or they will drop the message. 
   - you therefore can't support a signed system with any unsigned or incorrectly signed system on a particular channel.
2. But can you support _unsigned_ MAVLink 1 and _unsigned_ MAVLink 2 on the same channel? The answer is "kind of" but you shouldn't need to
   - A MAVLink 1 system can only receive MAVLink 1 messages, but a MAVLink 2 system can receive MAVLink 1 or 2 messages and can send either MAVLink 1 or 2 messages at a time. So in theory you could send those MAVLink 1 messages to just the MAVLink 1 target and otherwise work in MAVLink 2. 
   - IN practise almost all systems support MAVLink 2 now. The only things that should still be running MAVLink 1 are a few legacy mavlink peripherals, and these are usually set up to connect via a separate channel. This is why the handshaking is often setup to start at MAVLink 1 and switch to mavlink 2 as soon as you can negotiate it. That way the channel will stay MAVLink 1 for the few peripherals, and otherwise quickly switch to MAVLink 2. 

From the above there is no viable strategy if you're just using one channel. It should however be possible to connect different systems in the same MAVLink network using separate channels (possibly bound to different or the same ports).

hamishwillee, 2018-12-20:
Closed by #148

closed 2018-12-20 by hamishwillee