mavlink/mavlink-devguide #152: How do we cancel a mission upload/download?

hamishwillee, 2018-12-13:
[Mission Protocol · MAVLink Developer Guide](https://mavlink.io/en/services/mission.html) does not explain how to cancel a partially completed download/upload.

For download everything is driven by the GCS/API and that can just stop getting items whenever it likes - ie cancelling is just stopping.

For upload of mission to vehicle the API initiates the upload, which starts the process of the drone requesting mission items (which are returned in MISSION_ITEM or MISSION_ITEM_INT). In this case there is no ACK in the sequence, so how does an API tell the drone that it is ignoring future mission item requests and stopping the upload?

@julianoes , any ideas?

hamishwillee, 2018-12-14:
Further to this, 

1. What is the state of the mission during upload?
  - If I start uploading items, can the mission be running?
2. Is the update robust? i.e. if I update the mission and then cancel it or stop supplying items, is the mission reverted to the original set, or does it have some of the items from the original set and some from the new set?
3. When a message is finished, is there a state machine that says the mission is no longer running? ie currently our docs talk about missions running to a pause or to completion, but our fail states sometimes talk about a mission being "as if complete". That is fine, except that we don't define what that state means. I assume it means "mission has been running and has either completed or failed. It is no longer running and needs to be restarted/reset to run again.". We should define what conditions that requires.

julianoes, 2018-12-17:
> 1. What is the state of the mission during upload?
>    - If I start uploading items, can the mission be running?

This is how it works for PX4: There we have to storage places for missions, one is the current mission potentially being executed, and a second one for an ongoing transfer. The controller only switches over to a new mission if the mission has successfully been fully transferred. Therefore, if anything happens during the transfer like an error or a timeout, the controller won't use the uploaded items ever. For the case of cancellation, the SDK sends a `MISSION_ACK_DENIED` meaning that the mission transfer has been denied.

> 2. Is the update robust? i.e. if I update the mission and then cancel it or stop supplying items, is the mission reverted to the original set, or does it have some of the items from the original set and some from the new set?

Yes, it is (and should be) reverted at what it was. Or an alternative implementation would end up with no mission at that point and just have to loiter.

> 3. When a message is finished, is there a state machine that says the mission is no longer running? ie currently our docs talk about missions running to a pause or to completion, but our fail states [...]

The SDK uses the last "current mission item" and checks it again "total mission items" and if `current + 1 == total` it usually means we're done. Although it can get tricky for things like automatic RTL after mission. For that case, is the mission finished when RTL is triggered or is RTL then part of the mission.
What I would prefer and what might happen is that we switch to LOITER (or RTL) after a mission is done.

> 4. Is a partial mission upload allowed - ie i just send one item to the autopilot?

It's not implemented for PX4 as far as I know.

hamishwillee, 2018-12-20:
Thanks @julianoes !

hamishwillee, 2018-12-21:
How does PX4 handle invalid items - ie if a mission item wasn't supported? Would it still be accepted but just skipped?

julianoes, 2018-12-21:
I don't know, let's see:
https://github.com/PX4/Firmware/blob/fce35ba9d27c33cc65e1215682785debbd99d361/src/modules/mavlink/mavlink_mission.cpp#L1483
https://github.com/PX4/Firmware/blob/fce35ba9d27c33cc65e1215682785debbd99d361/src/modules/mavlink/mavlink_mission.cpp#L1054-L1065
Looks like we nack the transfer.

hamishwillee, 2018-12-21:
@julianoes Stop pretending you don't know everything :-)

julianoes, 2018-12-21:
I don't, I only sometimes know where to look for.

hamishwillee, 2018-12-24:
:-). Does PX4 fully support the geofence and rally point protocol too? MY understanding is that geofence is supported, but not rally points.

julianoes, 2018-12-26:
Ha! I don't know and looking at the code won't give a conclusive answer.

hamishwillee, 2018-12-28:
:-)

OK, so re cancellation:

> For the case of cancellation, the SDK sends a MISSION_ACK_DENIED meaning that the mission transfer has been denied.

This works for downloading, because PX4 will set system back to idle: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_mission.cpp#L625 (and do warning)

However for uploading to drone the PX4 understands that an early ACK is a fail case (https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_mission.cpp#L641) and reports a warning, but it doesn't cancel the expected upload. It will (based on inspection) basically wait for a bit and then start resending MISSION_REQUEST_INT for a number of retries.

So is that tested?

hamishwillee, 2018-12-28:
Also, I think perhaps we should properly support the cancel case with a cancelled enum value?

hamishwillee, 2018-12-31:
Great. I've created https://github.com/mavlink/mavlink/pull/1044 for a proper error report, after which "someone" will be able to fix up PX4.

hamishwillee, 2020-02-18:
Fixed by 1044

closed 2020-02-18 by hamishwillee

julianoes, 2020-02-19:
@hamishwillee someone was eventually me: https://github.com/PX4/Firmware/pull/14139 :smile: