mavlink/mavlink-devguide #297: is Image Transmission Protocol document outdated?

mnumanuyar, 2020-12-09:
in the document it says 

> ... one must set the following fields in the MAVLink message:
 ... 
freq: bigger than 0 for "frames per seconds", lower than 0 for "seconds per frame"

However there is no field for freq in  [DATA_TRANSMISSION_HANDSHAKE](https://mavlink.io/en/messages/common.html#DATA_TRANSMISSION_HANDSHAKE) definition.

In addition there is a note that starts with "At time of writing (March 2018)." either the date or the content should be updated.

Bug Page: [Image Transmission Protocol · MAVLink Developer Guide](https://mavlink.io/en/services/image_transmission.html)

hamishwillee, 2020-12-09:
Yes, this is screwed up. Looks like it has been for some time https://github.com/mavlink/mavlink/issues/691,  https://github.com/mavlink/mavlink/issues/1217
I'll look into it. Thanks.

closed 2020-12-09 by hamishwillee

hamishwillee, 2020-12-24:
Just to close the loop, I have updated the docs. 

For anyone else coming here the protocol is not intended as a general-use mechanism for image transfer.  Generally in most cases there are better ways to transfer images - mavftp, rtsp streams etc. It is still in use for sending images from PX4 flow to QGC to enable the flow unit to be focussed from the GCS, but that is about it.