mavlink/mavlink-devguide #301: FTP protocol incompatibilities

durka, 2020-12-10:
The documentation disagrees with PX4 and AP on the precise meaning of CreateFile and OpenFileWO. Unfortunately, the implementations are both more liberal than the spec but in different ways, so it'd be a breaking change to fix them.

| Method | O_CREAT | O_EXCL | O_TRUNC | O_WRONLY |
| --- | --- | --- | --- | --- |
| CreateFile | AP, PX4, docs | PX4, docs | AP | AP, PX4 |
| OpenFileWO | PX4 | | | AP, PX4, docs |

The disagreement is which FTP operations imply O_CREAT and O_EXCL. Also, the O_TRUNC behavior is inconsistent and not specified at all in the guide as far as I can see.

hamishwillee, 2020-12-10:
The PX4 implementation came first, but either way the spec needs to be tidied, and earlier break is better. @julianoes @MatejFranceskin Can one of your review this and suggest options?

julianoes, 2020-12-11:
I'm out, sorry.

dagar, 2020-12-15:
Is it preferable that opening a file write only (OpenFileWO) doesn't create it (O_CREAT)? At a glance I don't object to changing that in PX4.

How would you propose fixing O_TRUNC?

hamishwillee, 2020-12-15:
Truncate has some docs here: https://mavlink.io/en/services/ftp.html#truncate-file and https://mavlink.io/en/services/ftp.html#opcodes
There is also an implementation of FTP in MAVSDK. 

Docs indicate that for 
- createFile the file must not exist but all parent directories must exist
- for openfileWO the file must exist.
- for truncate just says truncate file at  `path` to `offset` length.

Looking at this, it looks like both AP and PX4 always create the file if it does not exist? Which is incorrect w.r.t. docs. @dagar What's best here - update docs to reflect reality for these two stacks or get stacks to do the right thing. 

The spec doesn't say anything about exclusive access. It makes sense that if you're opening for any operation that changes the file you really must have exclusive access. Does that need to be spelled out?

durka, 2020-12-16:
From my perspective, I was trying to find the right way to upload a file, whether or not it exists already. Per the docs, you can't do this -- need a round trip to see if it exists and decide whether to CreateFile or OpenFileWO. On ArduPilot, you can use CreateFile to do this, but not OpenFileWO as it would fail if the file didn't exist. However, on PX4 CreateFile would fail if the file *does* exist so you would use OpenFileWO for the same operation.

durka, 2020-12-16:
There are docs for the TruncateFile operation, but what I meant is it's not documented whether OpenFileWO truncates.

hamishwillee, 2020-12-16:
> There are docs for the TruncateFile operation, but what I meant is it's not documented whether OpenFileWO truncates.

You can open a file using OpenFileWO and write a bunch of data to it at different offsets. I think you're asking whether the open operation is an "overwrite" of the existing data or a "complete replace". 
- If it is a replace then the new file replaces the old file completely, and if it happens to be smaller, the old file will effectively have been truncated. 
- However if it is an over-write, in theory you could put chunks in the middle or at the end. In this case I'd assume you'd be able to increase the file size but not truncate it.

Is that a reasonable explanation of the problem/question?

[![](https://mermaid.ink/img/eyJjb2RlIjoic2VxdWVuY2VEaWFncmFtO1xuICAgIHBhcnRpY2lwYW50IEdDU1xuICAgIHBhcnRpY2lwYW50IERyb25lXG4gICAgTm90ZSByaWdodCBvZiBHQ1M6IENyZWF0ZSBmaWxlIG9uIERyb25lXG4gICAgR0NTLT4-RHJvbmU6ICBPcGVuRmlsZVdPKCBkYXRhWzBdPXBhdGgsIHNpemU9bGVuKHBhdGgpIClcbiAgICBEcm9uZS0tPj5HQ1M6IEFDSyggc2Vzc2lvbiAsIHNpemU9MCApXG4gICAgTm90ZSByaWdodCBvZiBHQ1M6IFdyaXRlIGRhdGEgaW4gY2h1bmtzPGJyPnVudGlsIGNvbXBsZXRlXG4gICAgR0NTLT4-RHJvbmU6ICBXcml0ZUZpbGUoc2Vzc2lvbiwgc2l6ZT1sZW4oYnVmZmVyKSwgb2Zmc2V0LCBkYXRhPWJ1ZmZlcilcbiAgICBEcm9uZS0tPj5HQ1M6IEFDSyggc2Vzc2lvbiwgc2l6ZT0wIClcbiAgICBOb3RlIHJpZ2h0IG9mIEdDUzogQ2xvc2Ugc2Vzc2lvblxuICAgIEdDUy0-PkRyb25lOiAgVGVybWluYXRlU2Vzc2lvbihzZXNzaW9uKVxuICAgIERyb25lLS0-PkdDUzogQUNLKCkiLCJtZXJtYWlkIjp7InRoZW1lIjoiZGVmYXVsdCJ9LCJ1cGRhdGVFZGl0b3IiOmZhbHNlfQ)](https://mermaid-js.github.io/mermaid-live-editor/#/edit/eyJjb2RlIjoic2VxdWVuY2VEaWFncmFtO1xuICAgIHBhcnRpY2lwYW50IEdDU1xuICAgIHBhcnRpY2lwYW50IERyb25lXG4gICAgTm90ZSByaWdodCBvZiBHQ1M6IENyZWF0ZSBmaWxlIG9uIERyb25lXG4gICAgR0NTLT4-RHJvbmU6ICBPcGVuRmlsZVdPKCBkYXRhWzBdPXBhdGgsIHNpemU9bGVuKHBhdGgpIClcbiAgICBEcm9uZS0tPj5HQ1M6IEFDSyggc2Vzc2lvbiAsIHNpemU9MCApXG4gICAgTm90ZSByaWdodCBvZiBHQ1M6IFdyaXRlIGRhdGEgaW4gY2h1bmtzPGJyPnVudGlsIGNvbXBsZXRlXG4gICAgR0NTLT4-RHJvbmU6ICBXcml0ZUZpbGUoc2Vzc2lvbiwgc2l6ZT1sZW4oYnVmZmVyKSwgb2Zmc2V0LCBkYXRhPWJ1ZmZlcilcbiAgICBEcm9uZS0tPj5HQ1M6IEFDSyggc2Vzc2lvbiwgc2l6ZT0wIClcbiAgICBOb3RlIHJpZ2h0IG9mIEdDUzogQ2xvc2Ugc2Vzc2lvblxuICAgIEdDUy0-PkRyb25lOiAgVGVybWluYXRlU2Vzc2lvbihzZXNzaW9uKVxuICAgIERyb25lLS0-PkdDUzogQUNLKCkiLCJtZXJtYWlkIjp7InRoZW1lIjoiZGVmYXVsdCJ9LCJ1cGRhdGVFZGl0b3IiOmZhbHNlfQ)

durka, 2020-12-17:
Yes I think so -- so OpenFileWO doesn't truncate. There's no distinction between "open for writing"/"open for appending" like in the standard library, because you send the offset with every write operation anyway.

And it doesn't really matter whether CreateFile truncates, because per the docs it's not even supposed to succeed if the file doesn't exist.

What do you think about my original question that led me to this -- the "upload a file in the fewest number of steps" use case?

hamishwillee, 2020-12-17:
@durka 
- I'm not sure about the _intent_ either - so I will ask about this in the [dev call next Wednesday](https://github.com/mavlink/mavlink/wiki/20201223-Dev-Meeting). That has people in it from ArduPilot and PX4 so I can hopefully co-ordinate a response.

Note, I'm not a technical expert - I'm a tech author and kind of "team co-ordinator".

LorenzMeier, 2020-12-23:
@durka Thanks for raising this. I've updated the MAVLInk docs after a discussion on the dev call:
https://github.com/mavlink/mavlink-devguide/blob/master/en/services/ftp.md

We're now matching the behavior of https://pubs.opengroup.org/onlinepubs/9699919799/functions/creat.html in the file creation (O_TRUNC) and we've clarified that the file needs to be created for OpenFileWO if it did not already exist.

PX4 PR: https://github.com/PX4/PX4-Autopilot/pull/16442

With these changes the FTP protocol spec is clear and any deviations from the spec will need to be addressed by the respective adopters (as I changed the implementation for PX4 to reflect the spec change we did today).

closed 2020-12-23 by LorenzMeier

davidbuzz, 2021-01-20:
So... we think following this last comment by lorenz and the px4 change/s that ardupilot and px4 are now both compatible and compliant. If there is any ground-station/s etc that were previously having compatability issues, they should review their codebase is compatible with the recent px4 changes ( matching ardupilot's functionality) ?

LorenzMeier, 2021-01-20:
@davidbuzz APM still would need to implement O_CREAT in OpenFileWO, which should however be a non-breaking change.

davidbuzz, 2021-01-20:
my reading of the ardupilot code is that it Ardupilot already does O_CREATE:  https://github.com/ArduPilot/ardupilot/blob/89f0ed2f8f262856d86e0bd8b53f61e99be0dfb9/libraries/GCS_MAVLink/GCS_FTP.cpp#L364   
``
                        ftp.fd = AP::FS().open((char *)request.data,
                                               (request.opcode == FTP_OP::CreateFile) ? O_WRONLY|O_CREAT|O_TRUNC : O_WRONLY);
```

LorenzMeier, 2021-01-20:
Ok, perfect, then that table above is outdated or was not correct.

durka, 2021-01-20:
The way I read that ternary expression it's applying O_CREAT for CreateFile but not OpenFileWO.

tridge, 2021-01-20:
> The way I read that ternary expression it's applying O_CREAT (and O_TRUNC) for CreateFile but not OpenFileWO.

right, which matches posix open with O_WRONLY. I note the ftp devguide says:
[image]
which does imply O_CREAT, but it is a bit dubious, as it doesn't mention O_WRONLY which is needed to be able to write.
I'm certainly happy to change ArduPilot to add the O_CREAT in this case if that is what is needed for clients. Exactly matching posix isn't the priority

julianoes, 2021-02-03:
@davidbuzz 

> my reading of the ardupilot code is that it Ardupilot already does O_CREAT: https://github.com/ArduPilot/ardupilot/blob/89f0ed2f8f262856d86e0bd8b53f61e99be0dfb9/libraries/GCS_MAVLink/GCS_FTP.cpp#L364

I think it should be `O_WRONLY|O_CREAT` for OpenFileWO as @durka writes as well.

---

I will follow up what MAVSDK does.

julianoes, 2021-02-03:
`O_TRUNC` added to the implementation in MAVSDK: https://github.com/mavlink/MAVSDK/pull/1306.