mavlink/mavlink-devguide #392: Is there documentation or example on how to use the C++11 generated libraries?

aarondewindt, 2021-09-18:
I have an incoming and outgoing stream of bytes which I need to serialize and deserialize to/from `mavlink::Message` objects. How do I do this? I can't find the documentation for the C++11 libraries or some example. So does anyone know of some example code doing this?

aarondewindt, 2021-09-18:
So to be more specific. I want to deserialize the stream into messages of their own type, but return a `std::shared_pointer<mavlink::Message>`.

hamishwillee, 2021-09-19:
@aarondewindt The only examples and docs we have are for [C and Python](https://mavlink.io/en/getting_started/use_libraries.html). @julianoes Can you advise?

julianoes, 2021-09-20:
What about MAVROS? Does that help?
e.g. https://github.com/mavlink/mavros/blob/ros2/mavros/src/plugins/manual_control.cpp

vooon, 2021-09-20:
@julianoes i think plugins code didn't help here, because mavlink transcoding is hidden by the api.

@aarondewindt C++11 uses C library to finalize serialization and message parsing.
You should implement `mavlink::mavlink_get_msg_entry()`, as i gives extra crc byte.

https://github.com/mavlink/mavros/blob/87f2b114206057595ef4ed72e75057aed05e71cb/libmavconn/src/mavlink_helpers.cpp.em#L68-L78

To serialize mavlink::Message you need to:
1. Copy your message to mavlink_message_t trough mavlink::MsgMap by calling Message::serialize();
2. Set headers and crc by mavlink_finalize_message_buffer();
3. Copy mavlink_message_t to bytes buffer (which then you could send)

https://github.com/mavlink/mavros/blob/87f2b114206057595ef4ed72e75057aed05e71cb/libmavconn/include/mavconn/msgbuffer.hpp#L60-L78

To parse you should call mavlink::mavlink_frame_char_buffer() and then use MsgMap with resulting mavlink_message_t to be able to Message::deserialize().

https://github.com/mavlink/mavros/blob/87f2b114206057595ef4ed72e75057aed05e71cb/libmavconn/src/interface.cpp#L101-L125

https://github.com/mavlink/mavros/blob/87f2b114206057595ef4ed72e75057aed05e71cb/mavros/include/mavros/plugin.hpp#L161-L174

julianoes, 2021-09-20:
@vooon sorry, you're right. I did not read the question correctly :man_facepalming:.

aarondewindt, 2021-09-20:
@vooon Thaaanks, that help quite a bit. I figured I had to use the C library and `mavlink::MsgMap` but implemented it using `mavlink_parse_char` instead, I haven't had a chance to test it though. I'll post an short update once I get it to work. Maybe with a gist as a simple example for future reference.

hamishwillee, 2021-09-20:
A simple example or advice we can add to the docs would be great.